cta-observatory/ctapipe #2830: Handling of random seeds

kosack, 2025-09-04:
in #2819 (and before), we have a few places where the random seed is hard-coded to `0`, which means the random sequence is always the same.  

It may be better to allow a configurable random seed at the Tool level perhaps, e.g. be able to set "MyTool.random_seed=x", where the default can be 0., but where we can change it when necessary.

maxnoe, 2025-09-04:
We do allow that for the rngs where this might be important, e.g. the seeds to sample events for training or to add noise.

We currently do not for the seed in the image parameters fitting that choses which pairs of pixels are used for the timing fit.

I really don't see a use case where a user would want to adjust that seed.

kosack, 2025-09-04:
Ok, if the imporant cases are already handled (I didn't check I admit), then I think it's ok. I'll close this.