cta-observatory/ctapipe #2839: Event IDs duplicated/mismatched between tables when writing simtel to HDF5 with ctapipe-processor

jbuces, 2025-09-18:
**Describe the bug**
When writing an NSB simtel file to hdf5 format using the ctapipe-processor (v 0.23.2), the /r0/event/telescope/tel_001 and  /simulation/event/subarray/shower tables  preserve correct event_ids. But some tables as  /dl1/event/telescope/parameters/tel_001 or /dl1/event/telescope/images/tel_001 contain duplicated/misaligned event_ids, with many entries appearing twice. The indexing of the mistaken tables are also different between them. The problem is not reordering inside the table but actual duplication.

**To Reproduce**
Steps to reproduce the behavior:
1. Processing with sim_telarray the file dummy100000.corsika.
2. Writing the simtel file to HDF5 with the ctapipe-processor.

**Expected behavior**
Having all tables with the same and correct event_ids.

**Supporting information**
The /dl1/event/subarray/trigger table:

[image]

The /dl1/event/telescope/images/tel_001 table:
[image]

The /dl1/event/telescope/parameters/tel_001 table:

[image]

The command used is:
        cmd = [
             f"ctapipe-process",
             "-v",
             "--overwrite",
             f"--input={file}",
             f"--output={output_file}",
             f"--config={args.config}",
             f"--DataWriter.write_dl1_images=True",
             f"--DataWriter.write_dl1_parameters=True",
             f"--DataWriter.write_dl2=False",
             f"--DataWriter.write_r0_waveforms=True",
             f"--DataWriter.write_r1_waveforms=False",
             "--ProcessorTool.provenance_log=...",
             f"--DataWriter.write_index_tables=True"
            ]

**Additional context**
The simtel files are being processed with a SiPM camera configuration, but this or other parts in the process should not be relevant since all the gamma files have been processed in the same way and do not suffer from this duplication.

maxnoe, 2025-09-18:
I don't see any duplicates in the screenshots you have posted.

Could you show an actual error and provide us with an example file?

Maybe a small one that reproduces the issue and could be added to the test datasets?

jbuces, 2025-09-18:
One of the problems is that no error showed up in the log while processing, I realised this when checking some results.

Just by looking at the event_id column in /dl1/event/telescope/parameters/tel_001 for example and checking for duplicates with:

values, counts = np.unique(tab["event_id"], return_counts=True)
for v, c in zip(values, counts):
    if c > 1:
        print(f"Value {v} appears {c} times")

The prints show:

Value 100 appears 2 times
Value 200 appears 2 times
Value 300 appears 2 times
Value 400 appears 2 times
Value 500 appears 2 times
Value 600 appears 2 times
Value 700 appears 2 times
.....(up to)
Value 4294300 appears 2 times
Value 4294400 appears 2 times
Value 4294500 appears 2 times
Value 4294600 appears 2 times

It restarts the count at a given range:

np.where(table["event_id"]==100)[0]

array([    0, 57054])

The behaviour is a bit different for the /dl1/event/telescope/images/tel_001 table, the ones repeated are the last computed indices:
Value 9960100 appears 2 times
Value 9960200 appears 2 times
Value 9960300 appears 2 times
Value 9960400 appears 2 times
Value 9960500 appears 2 times
Value 9960600 appears 2 times
(up to)
Value 9969600 appears 2 times
Value 9969700 appears 2 times
Value 9969800 appears 2 times
Value 9969900 appears 2 times
Value 9970000 appears 2 times

The tables under simulation and r0 are correct.

What file should I pass? The h5 with the 100 000 events is very heavy I could generate one corsika with fewer events and process it but maybe when writing it into the hdf5 format it doesn't reach this 'limit'. I will try. The simtel file with the 100 000 events has a correct indexing of the events.

maxnoe, 2025-09-19:
How large is the simtel file?

Can you share it using some cloud storage or by uploading it to the LST IT cluster?

jbuces, 2025-09-19:
The simtel file is 22 GB and the h5 file is 26 GB, I have uploaded both of them to the IT in /fefs/aswg/workspace/jorge.buces/nsb_triggerless_bug . Let me know when you have downloaded them so that I can delete them and not take up so much space.

maxnoe, 2025-09-19:
Just to be sure: this is from a single run and not a merged or appended to file?

jbuces, 2025-09-19:
Yes, I processed a single corsika file dummy100000.corsika , and then wrote only this output simtel with the ctapipe tool.

TjarkMiener, 2025-09-19:
For the sake of completeness, I'm attaching a screenshot with similar behaviour:

[image]

maxnoe, 2025-09-19:
I just ran over the simtel file with ctapipe 0.26 and cannot reproduce the issue:

```
In [13]: from ctapipe.io import read_table

In [14]: parameters = read_table("workspace/jbuces-ctapipe-issue/test.h5", "/dl1/event/telescope/parameters/tel_001")

In [15]: len(np.unique(parameters["event_id"]))
Out[15]: 100000
```

kosack, 2025-09-22:
I don't see any changes that would affect any code where events are written. I was worried a bit that it could be due to appending to an old file, but we don't use append mode and if you use `--overwrite` the original file is actually explicitly _deleted_ before the writing starts.  

Could you maybe also post how you are _reading_ the table? Maybe there is where the issue lies?

jbuces, 2025-09-24:
I don't know if it is relevant but I am running in a machine with Scientific Linux 7.9

jbuces, 2025-09-30:
Hi, when putting the flag write_index table to false, the duplication no more happens, so maybe the error comes when reading the table with the Pytables index generated.

maxnoe, 2025-09-30:
mmh... we don't really use that option in production and there is no code in ctapipe that relies on these index tables being present.

I'd recommend to switch it off. I'll have a look why this creates this bug, but I'll probably just open a PR to entirely remove this option.

maxnoe, 2025-10-07:
I reproduced the error: when `--write-index-tables` is given, I get duplicated event_ids.

It's not code in ctapipe though. The data is really like that in the file, even low-level access with tables shows the wrong data.

I can only conclude that the generation of table indices somehow corrupts the file.

TjarkMiener, 2025-10-07:
So `--write-index-tables` should be ditched?